smartavionics/Cura #256: Infill layer thickness removes skin edge support

PhilBaz, 2024-07-08:
**Application version**
(The version of the application this issue occurs with.)

current

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

win 10

**Printer**
(Which printer was selected in Cura?)
Voron 2.4

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Hi again!

I'm using infill layer thickness to double up the thickness--half the number of layers that need to be printed.

It works as expected except for the fact that it essentially eliminates all but one skin edge support path. With irregular shapes I generally use 6-9 edge layers to ensure the final layer is supported with sparse infill settings.  In the project Im slicing now its killing the top skin quality.

**Actual results**
(What happens after the above steps have been followed.)

skin edge support lines are excluded

**Expected results**
(What should happen after the above steps have been followed.)

Skin edge support lines should remain.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

Let me know if you want one. It's not really file specific...I think.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Perhaps the lines could be included in the doubled up layers? 

Another solution might be to exclude the final n# of layers from adhering to the infill layer thickness setting (in the infill section) so that the default layer thickness is used--I'm assuming this would reincluded the skin edge support layers? With the latter approach perhaps it could be tied directly to the number of skin edge support layers?

Thanks again Mark!

smartavionics, 2024-07-12:
Hi, I've done this, not exhaustively tested but it will be in the next release.

PhilBaz, 2024-07-12:
@smartavionics 
Ahhh, Thank you Mark! Much appreciated :)