smartavionics/Cura #257: Layer view color switches suddenly after one too many layers is shown

Botspot, 2024-08-17:
**Application version**
Latest release aarch64 appimage downloaded by Pi-Apps

**Platform**
Raspberry Pi 5 running 4k pagefile kernel

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. Slice something tall or complex.
2. Slide up/down the layers and watch for the appearance to change suddenly with one more layer being added.

**Screenshot(s)**
1. Layer 475:
    [image]
3. Layer 476:
    [image]

**Actual results**
Color and contrast suddenly changes.

**Expected results**
Layers appear the same regardless of how many are shown at once.

**Additional information**
This seems to happen at a different layer number, depending on the model. That leads me to believe that this is something that triggers after a certain number of triangles are being rendered.

Botspot, 2024-08-17:
I am hoping this is easy for you to reproduce. Otherwise, I will need to dig into it more and see what is unique about my setup, or just decide it's not a big deal and close the issue to save some effort.

smartavionics, 2024-08-17:
This artefact is an unfortunate side effect of how I have implemented the layer view for the Pi. Due to the fact that the Pi has a rather low power GPU compared to your average PC, I have made two versions of the shader used to compute the layer view lines. There is a shader that is similar to the shader used for desktop OpenGL that UM Cura uses. That displays each gcode line as a 3d tube made with 4 sides so it has a squashed rectangle cross section. I also provide a simpler shader that displays each gcode line as a 2d plate that is angled towards the viewer. This much simpler shader executes quite a bit quicker than the 3d alternative and so when the layer view contains a lot of elements, I use the 2d shader. What you're seeing is the transition between the two shaders.

The threshold is 500000 elements, you can actually change that by setting an environment variable, CURA_MAX_LAYER_VIEW_3D_ELEMENTS, to a different number.

Botspot, 2024-08-17:
Good to know. If that was documented somewhere, I apologize for not looking.

Would there be any way to improve the 2d shader so that some angular contrast can remain? As of now, once the 500000 elements limit is exceeded, it becomes harder to see any depth in the model.

smartavionics, 2024-08-17:
I'll take another look at the 2d shader to see if I can improve that.

smartavionics, 2024-08-17:
There's also another hack in the 2d shader to increase the performance. If you zoom in to the model do you find that once you get closer than a certain distance, the contrast improves?

You can get three "qualities" of shading:

With less than 500000 elements, all shading is done with the 3d shader and zooming in/out should make no difference.

With more than 500000 elements, the 2d shader is used and if zoomed out sufficiently, an even simpler shader algorithm is used. Unfortunately, at this time, there's no option to disable that behaviour.

I think from your image above that it is the zoomed out case that needs improving.

Botspot, 2024-08-17:
Yes, there is a sudden change from zooming in.

Before:
[image]
After:
[image]

I do want to note that this 2d shader of yours does not look too bad when the model is viewed from an elevated position. Angular contrast is still nonexistent, but the color is not supersaturated as it is when viewing the model from beneath.

Botspot, 2024-08-17:
This is the 3d shader from the same vantage point. It is quite similar to your zoomed-in shader, but the blue arrows I've added show areas that better show their more shallow angle from the perspective of the camera.
[image]

smartavionics, 2024-08-19:
I have made some changes to the shaders. There is now less visual difference between the 3d shader and the 2d shader. I think the contrast when zoomed out using the 2d shader is better. It's never going to be perfect.

I have also added another environment variable that lets you control the shader switch when zooming out. It's called CURA_LAYER_VIEW_RESOLUTION. If not set, the behaviour is unchanged, i.e. it will use a lower resolution shader when zoomed out. If set to 1 it will always use the higher resolution 2d shader irrespective of the zoom. When set to 0 it will always use the lower resolution 2d shader irrespective of the zoom.
If you wish to try the changes you can by following these steps...

1) extract the contents of the Cura appimage by doing:
```Cura-4.20.23-mb-aarch64.AppImage --appimage-extract```
This will create a directory called squashfs-root

2) Unzip [shader-mods.zip](https://github.com/user-attachments/files/16657621/shader-mods.zip) and that will replace the files that have been modified.

3) Run the modified cura by executing squashfs-root/usr/bin/cura.sh

4) let us know what you think. If you want to have a fiddle with the shaders, feel free. Typing ctrl-. in the layer view causes the shaders to be reloaded so you don't have to restart Cura every time you edit the shader file.

smartavionics, 2024-08-19:
Oops, I missed a file from the zip archive, here's a new one...

[shader-mods.zip](https://github.com/user-attachments/files/16660109/shader-mods.zip)