smartavionics/Cura #258: Retract while Hop instead of retract and then hop.

McNugget6750, 2024-10-04:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
Instead of retract and then hop (z move up then travel) it would be great to retract and z move up at the same time and then travel. Not sure if this would result in lots of little antennas but it would be faster and leaves less time for the nozzle to make a mark on the top skin while the printer is waiting for the retract to finish

**Describe alternatives you've considered**
making the retract minimum travel larger but I don't think that will result in the desired outcome. I'd still end up with little imprints over my top skin.

**Affected users and/or printers**
Everyone who would like to activate this feature.

**Additional context**
I found myself with a great looking surface but if I wanted to slide it over a mousepad it would rip out several little threads....

smartavionics, 2024-10-06:
This may well best be done using a post-processing script to combine the retraction and z-hop moves. I should think that @gregvaliant could say whether this is easy to achieve or not as he has much experience in that area.

GregValiant, 2024-10-06:
Hello.
I have found that a quick pass or two with a 10" fine-cut mill file gets rid of the little pips quite nicely.  Sometimes a print needs some benchwork.

I have a post processor that will add Z-hops based on the travel distance rather than whether there is a retraction or not.  A side effect is to divorce the Z-hop from the retraction.  You can turn Z-hops off in Cura, and set the retraction settings to what works best and enable that post processor (which looks at the distance of each travel move).  So if it is set up to hop prior to any move over 12mm then that's what it will add whether or not there was a retraction.  I only use Z-hops when there is a chance that a feather edge is going to develop and there is a good chance that the nozzle is going to whack it during a travel.  So another setting is to tell the script what the layer range is that Z-hops will be required on.

This request though is slightly different.  It would not be a problem to combine the Z-hop with the retraction.  Looking at just the timing (and ignoring acceleration) Z-speed is much slower than retraction speed _BUT_ generally the distance is a lot less.  The printer will see a line like "G1 E-7.0 Z0.5 F3500" and calculate the steps that need to be sent so both motors arrive at the end position at the same time.

Just thinking out loud here and considering a bowden printer...
G1 E-7 F2100
will take 0.2 seconds to complete.
G1 Z0.5 F600
will take 0.05 seconds to complete.
If they are consecutive the two moves will take 0.25 seconds.  Since the "retract-hopup" is always coupled with a "prime-hopdown" and if there are 7000 of them in a file then yes, it's going to add nearly an hour to the print time.

If it's a DD printer then the retraction distance is a lot shorter.
G1 E-2 F2100
will take 0.06 seconds to complete.  That makes the total time for 1 event 0.11 seconds which is about 1/2 the time. 

In either case, it's the retraction that will be the deciding time factor since the Z motor will have no trouble keeping up.
In the case of the bowden printer you would save .05 seconds on each event (11.6 minutes less than the 60 minutes it was going to take).  In the case of the direct drive it would also be a saving of 11.6 minutes (11.6 less than the 30 minutes it would have taken).

So there is no great gain there and because you are lifting the nozzle without first relieving the pressure, I think you would have bigger zits and more stringing.

It wouldn't take much time to write a script that picks out the retraction-hop combinations and makes then into a single line, but I'd probably need to get it done before Milton comes calling and wipes out my house.  If I don't get it done before the house gets blown away you can still use a file.

GregValiant, 2024-10-06:
I don't have a lot of confidence that this will work as expected.  I'll post it on my [Github](https://github.com/GregValiant/GV-PostProcessors) page as well.  If there are problems let me know here or over there.
[CombineZhopsAndRetracts.zip](https://github.com/user-attachments/files/17270427/CombineZhopsAndRetracts.zip)

Sophist-UK, 2024-10-06:
Retract speed is important - if you combine the retract and the hop you would have to set the hop speed so that it completes at the same time as the retract.

Also I suspect that hopping as you retract might lead to worse blobs.

GregValiant, 2024-10-06:
The script uses the existing retract speed for the Z-hop.  It finds a zhop and then grabs the Z height and just adds it to the retract line.
Let's say it's the second layer at a 0.20 layer height and a 0.40 hop height (and relative extrusion) then:
**G1 F2100 E-7.0
G1 F600 Z0.6
M106 S255**
would become
**G1 F2100 E-7.0 Z0.6 ;  Combined Retract
M106 S255**
The Z-hop lines themselves just get deleted.

Going the other way (for a hop-down and prime situation)
**G1 F600 Z0.2
G1 F2100 E7.0
M106 S255**
becomes
**G1 F2100 E7.0 Z0.2 ;  Combined Prime
M106 S255**

I agree that it will probably be stringier with bigger zits.  I was thinking maybe a wipe move would be better but then it probably shouldn't be combined with the retraction because the pressure in the nozzle really needs to be lowered first.
BTW, I put switches in so the user can select to alter both the retract AND primes, or just one or the other.
[image]

I like my bastard file though.  It's simple, easy to understand, and I can use it for other things.

McNugget6750, 2024-10-07:
Wow! This is an interesting script! I'll take a look!

I was able to solve most of the issues by increasing the minimum distance needed to perform a retract / hop in the first place. The default was 0.5 or 0.8mm which is essentially nothing. Since I was trying to tune the new single wall top skin feature, it was retracting all the time leaving these tiny blobs over all the features of the print.

Increasing the minimum distance to 2mm totally changed this and the surface is now next to perfect. I love the new top skin settings as they really do make a huge difference on the Bambu Labs printers.

I spent a few days comparing the gcode that pops out of Bambu Studio at 0.2mm layer height with my custom printer settings on MB Cura and was able to replicate basically everything except for the speed. My older printer just can't do those but I don't mind. Now I have two printers printing essentially at identical very high quality.
The only setting I'm still calibrating is the infill overlap as I can see the infill pushing every so slightly into the perimeter which is only visible in at certain light angles. But I can see it so I'll put more effort into it.

One thing became very clear immediately: Input shaper and pressure advance have completely changed my printer and have given me a serious boost to try make it as perfect as the P1S is printing...

McNugget6750, 2024-10-07:
[PLA0.2mm.zip](https://github.com/user-attachments/files/17271900/PLA0.2mm.zip)

McNugget6750, 2024-10-07:
I'm running the script with the worse 0.8mm minimum travel right now - just to see what the result actually looks like.

smartavionics, 2024-10-07:
Hi Greg, sorry for the delay in responding, I've been working nights. Anyway, you appear to have this well under control. Thanks a lot for helping.

Sophist-UK, 2024-10-07:
I am now wondering whether it might make sense to break the retract into two pieces, doing the first piece alone and the second piece in conjunction with the hop. To use the same example as earlier:

Let's say it's the second layer at a 0.20 layer height and a 0.40 hop height (and relative extrusion) with:
G1 F2100 E-7.0
G1 F600 Z0.6
M106 S255

The retract will take 0.2s (F2100 is 2100mm/min or 35mm/s), and the hop will take 0.06s, so we can break this into two commands, the first a partial retract for 0.14s, and then the final part of the retract and the hop for 0.6s:
G1 F2100 E-4.9; Partial retract
G1 F2100 E-2.1 Z0.6 ; Combined hop and final retract
M106 S255

This will take the exact same time as the previous combined, hop and retract, but will do most of the retract in place, and only hop for the last part, and hopefully this will keep the zits pretty much the same whilst reducing the time.

McNugget6750, 2024-10-07:
my extremely limited experience with Bowden extruders doesn't really allow me to have an opinion on this.
My direct drive extruder only needs 0.5mm at 65mm/s retraction and a zhop of 0.2mm at 20mm/s. This is all very fast and I doubt splitting this command into two for DD is needed.

In fact, the part finished and the surface quality isn't any better than when first retracting and then hopping. For PLA I will therefore not use the script and rather increase the minimum distance traveled before retract feature to optimize the retractions away rather than optimizing retractions and hops.

For PETG, it might be worth a test as it needs much longer and much slower retractions to not cause stringing.

GregValiant, 2024-10-07:
Thanks Mark.  You're the first person who's ever had to deal with my code and said I had _anything_ under control.

It will work as advertised but I'm with @Sophist-UK ... I don't think it's really going to help.  I'm thinking "retract-wipe-lift" would be a lot better.  A problem is "which way to wipe" and that isn't easy to determine in post.  If the wipe was just 1/2 line width maybe it wouldn't matter.  If someone wanted a 1mm wipe then the vector would need to be determined and the wipe would back up along that vector.  Messy and slow.  But with a short wipe then each wipe could always be in the same direction (relative to the end of the extrusion) and the nozzle would (should) stay over the print.

This was a nice break from hanging hurricane windows and fooling with my generator.

Sophist-UK, 2024-10-07:
You can coast before the point of retraction so that the pressure can be allowed to reduce before the retraction / hop. Not sure whether that has the same effect as a wipe would have.

McNugget6750, 2024-10-07:
I feel like what you're describing is what Pressure Advance is for. It's so easy to calibrate and reduces the pressure consistently before the retract.

McNugget6750, 2024-10-07:
I completely turned off coasting and wipe, too, after calibrating Pressure Advance.
Wipe causes little streaks and coasting drastically depends on how fast you're moving the nozzle. So infill, walls, and perimeter will look entirely different with coasting. Plus, the seams will never look great not matter what.

With PA all of those issues fall away. Corners look great, seams look great, top surfaces look great - everything just works all of a sudden and all the workarounds I calibrated for months could be turned off.

What really helps is not leaving the nozzle on the surface. Hence the idea of hopping and retracting at the same time. But again, the test has confirmed that the little blips get even pointier and there is not a good idea.
Bambu uses really sharp nozzles that have a very pointy end without any flat part. Therefore their blips where the nozzle rests is tiny. Plus they move so fast that the plastic doesn't have time to get ironed / stamped in those spots. If I could get a hardened steel nozzle that fits my E3D Hemera that has a similar nozzle shape, I'd probably try it out

GregValiant, 2024-10-07:
I always use the SmartAvionics fork for spiralize and I always run the line width at 150% of the nozzle diameter.  It allows for much stronger prints.  I suppose if the nozzles were pointy I could just go up a size.  The 0.4 at a 0.6 line width works very well and the flat area around the nozzle opening does iron the extrusion and the next extrusion has a nice flat place to stick to.

McNugget6750, 2024-10-17:
After testing this, I don't think this is needed anymore. Thanks all, closing!

closed 2024-10-17 by McNugget6750