smartavionics/Cura #259: Cannot execute AppImage on Raspberry Pi 5

McNugget6750, 2024-10-04:
Downloaded, made executable, doesn't run:

> -rwxr-xr--  1 mcnugget mcnugget 220442884 Oct  3 18:36 Cura-4.20.24-mb-armhf.AppImage
> 
>  $ ./Cura-4.20.24-mb-armhf.AppImage
> bash: ./Cura-4.20.24-mb-armhf.AppImage: cannot execute: required file not found

It doesn't give me any other output.

> $ uname -a
> Linux dedsec 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt1 (2024-09-26) aarch64 GNU/Linux

McNugget6750, 2024-10-04:
> strace ./Cura-4.20.24-mb-armhf.AppImage 
execve("./Cura-4.20.24-mb-armhf.AppImage", ["./Cura-4.20.24-mb-armhf.AppImage"], 0x7fffff6bc4c0 /* 48 vars */) = -1 ENOENT (No such file or directory)
strace: exec: No such file or directory
+++ exited with 1 +++

smartavionics, 2024-10-04:
Hi,Just noticed, you're trying to run the 32 bit appimage on a 64 bit system. You need the arch64 appimage.

closed 2024-10-04 by McNugget6750