smartavionics/Cura #260: Slicer inconsistencies between identical layers

McNugget6750, 2024-10-09:
**Application version**
4.20.24-mb

**Platform**
Windows 10 x64

**Printer**
custom

**Reproduction steps**
load my project file and look at the sliced output in the preview

**Screenshot(s)**
[image]
[image]

**Actual results**
There are inconsistencies already visible in the sliced preview that would indicate this will also happen on the actual print. I have seen various artifacts on my prints like this but this is the first time I realize they are already visible in the slicer...

**Expected results**
A smooth surface that shows consistent layer lines

**Project file**
[MM_PCB Enclosure for a 78x54mm Board.zip](https://github.com/user-attachments/files/17309172/MM_PCB.Enclosure.for.a.78x54mm.Board.zip)

McNugget6750, 2024-10-09:
oh nice! Yet another setting I need to take a look at! Thanks, I'll report back!

smartavionics, 2024-10-09:
BTW, I'm glad you created this issue because while looking at the preview I discovered some problems with the narrow gap filling that was causing double extrusions in some places. I have now fixed that for the next release.