smartavionics/Cura #261: Different speeds for different overhang angles

McNugget6750, 2024-10-17:
Essentially, https://wiki.bambulab.com/en/software/bambu-studio/slow-down-for-overhang

I noticed that slowing down for outer walls really changes what the print looks like. Especially if the cooling system is sub-optimal. There is a general setting under experimental that allows for setting a fixed angle to go to a fixed (slower) speed but it's only one angle threshold, not a table and factor.

Factor could take just two values between which two speed settings are interpolated.
A table could take some four different angles / percentages and use the corresponding speed settings.

Or use an interpolation between the different values in the table to create a smooth transition between all speed settings and overhangs.

smartavionics, 2024-10-17:
I can't speak for UM Cura but my releases already vary the speed depending on the overhang angle as you can see in this image...

[image]

McNugget6750, 2024-10-17:
Ah! I guess I didn't see the color scheme page! I turned the setting on this morning, but I was expecting it to just switch to 50% speed at the angle that I set. If it's doing some linear speed reduction based on the angle, it wasn't obvious from the settings. I'll try this!

McNugget6750, 2024-10-17:
[image]
you are correct! Closing.

closed 2024-10-17 by McNugget6750