smartavionics/Cura #262: Extra wall underneath -first- top layer and treating -first- top layer like bridge

McNugget6750, 2024-10-23:
Bambu Studio adds one extra wall to provide improved adhesion for the first layer of the top surface.
[image]
[image]

They also treat the very first layer of top as bridge allowing to set bridge and top speeds separately, resulting is fewer rips in the first layer of the top layers.
[image]
[image]

I have tried to find settings to create the same in your Cura but was unsuccessful so far. Maybe you can point me in the right direction or consider adding this functionality in future builds?

smartavionics, 2024-10-23:
There isn't a setting to explicitly ask for an extra wall beneath the perimeter of the first skin layer but you will get those extra walls if you are using skin edge support.

As for printing the first skin layer using bridge settings, just set the Bridge Sparse Infill Max Density setting to a value greater than the infill density and it will do just that.

McNugget6750, 2024-10-23:
Oh sweet, I'll check both of these out and report back!

McNugget6750, 2024-10-23:
Both do what I needed! Thanks a lot for the support!

closed 2024-10-23 by McNugget6750