smartavionics/Cura #265: Infill - grid with last layer rectilinear

PhilBaz, 2025-03-27:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Yes, stronger infill types tend to cross. Others that are stronger but don't cross are slower. These crossing points can contribute to failed bridge sections, which in turn snowballs into surface blobs. If you'll recall, my white whale has been these surface blobs. I sell products with large flat surfaces and have spent much time and testing to minimize them. With ABS that is. Many of the requests iv made ( and you have graciously fulfilled) have been toward this same objective, 100% bridging reliability. 

Finally, I had the thought that It would be great to have the strength and reliability of grid but with the last layer as rectilinear. 

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Grid, & Hex triangle, with a final layer that does not cross. For rind that would simply be Rectilinear. For hex triangle it could simple be one of the 3 diagonals remains.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

A modifier but that's got problems I think, and would not be fun to do over and over..

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

anyone whos anal about surface quality.

**Additional context**
(Add any other context or screenshots about the feature request here.)

I'm on my knees, I'm begging please! Thanks again Mark.

smartavionics, 2025-03-27:
It seems to me that the skin edge support line distance setting (which you have tried) almost does the right thing except where the infill is completely covered by skin where it reverts back to the regular infill pattern and density because the skin's edges extend to the edge of the infill area or beyond.

So I suggest that we simply use the denser infill whenever the infill area has some skin above it irrespective of whether the skin completely covers the infill area or not. I think that would give you what you require.

I can also introduce a setting to let you specify the pattern used for the density infill.

Does this sound reasonable?

smartavionics, 2025-03-27:
On further thought, I am now sure that being able to specify the pattern for the dense infill helps as zigzag looks reasonable irrespective of what pattern is used below.

Zig-zag above grid ...

[image]

Zig-zag above tri-hex ...

[image]

PhilBaz, 2025-03-27:
Hey Mark!

I think the Zig zag above grid looks Perfect! As long as the zig zag pattern is always riding on top of the grid lines and then inside the boxes that should be perfect. 

Would this be in the bridging line types option section, and also be subject to the other bridging options?  The only problem would be with density since it should be a product of the geometry. Though some density options would be great, while respecting the underlying grid.

I think the hex triangle zig-zag above has the same problem as with standard bridging. It will hit the points where lines cross, making a failed connection much more likely, and then the snow ball begins. I think. If it was more sparse, and only rode on top of the lines it was parallel with, that would solve the problem. I think. I also suspect the tight corners of the triangles where it hits two lines almost immediately are as problematic as a crossing point..... I personally like hex-triangle infill. Maybe just me. 

I guess in a sense what zig-zag above grid does is marry the rectilinear pattern of the bridging to the grid infill? *** That gives me the idea that perhaps the bridging settings alone would be fine, as long as the bridge could be made to ride (rectilinear style) perfectly above the grid. I guess that's currently not possible (or maybe rather just fiddly) because bridge spacing is by density.

Alternately perhaps standard settings for bridging control that have some relationship to the grid infill pattern would be even more flexible? Line distance and x y offset? Also, I'm not sure how this works under the hood, does it take its coordinates from the underlying infill, so it would automatically ride along the top of a grid or hex triangle if the line distance were set according to the geometric relationship.... just thinking out loud here.

I should mention, after some testing of rectilinear infill pattern, it seams when a non-crossing infill patter is used that bridging speed almost doesn't matter for reliability. Can bridge at 200mm, no problem.

Excited to see where this goes! Thanks for indulging me yet again Mark! 

ps...As I've mentioned before. Your branch of Cura actually is the best and most capable slicer when real control is needed. 100%. I don't think people understand that. But what you can achieve with this slicer is unmatched elsewhere. Its not even close. On the flip side, main Cura development seems pretty non existent these days. They gave the world Arachne then basicaly gave up...at least it seems to a bystander like myself.  I use Orca for day to day stuff, and your slicer for my important and professional slicing, where quality matters above all else. The workflow and UI of Orca is very nice though, I must admit.

PhilBaz, 2025-03-27:
Yeah, boom! 

Never though of this before your zigzag over grind. Its like manual zigzag over grid using bridge settings.  What a pain in the ass that was!...hahah

Well Ill be curious to see what you think is the right solution here. 

Ill test this out, see how it goes.... I think were actually on to something here!  Bridging should use the underlying infill geometry for better surface quality. It makes sense!

[image]

smartavionics, 2025-03-27:
Hi Phil.

Yes, a problem with what I've tried so far is that it doesn't avoid putting the higher density infill on top of a crossing point on the layer below. I need to think more about this.

Now, one thing that is confusing me is that you keep referring to "bridging". To me, a bridge is where you are printing walls and skin unsupported with air below. You are using "bridging" differently here?

PhilBaz, 2025-03-27:
Yes! Those both look like excellent! Id be interested to try the first grid option you showed as well with the zigzag riding over top of the grid pattern. 

The issue is more when a bridge extrusion goes over a crossing point, of axis/pattern,
 as often happens with hex-triangles. (image). I don't think its an issue when riding atop an infill extrusion.

[image]

One thig id love from this solution is essentially to merge it with the bridge settings and give it the level of control found in bridging, fan speed, flow, speed, density, overlap, all in one place. Maybe that's way more work.  If these could be bridging line types that would be ideal from my perspective. Or maybe you are envisioning building up 'skin edge support density' as more of an 'internal bridging' feature as found in Orca. Either way, more control with 'internal bridging' is essential...as you can see by my years long battle.

PhilBaz, 2025-03-28:
Hey Mark,

I've been testing zig-zag over grid, using my little bridging hack, tuned to duplicate your first concept with the zig-zag riding atop the grid lines. 

It does indeed produce superior results. So far total reliability at much higher speed. Its a decided improvement. I've run about 4 rounds of testing now. 

Awesome.

smartavionics, 2025-03-28:
Hi Phil,

I can't look at this for a few days but I will put something together for you to try next week.

smartavionics, 2025-04-03:
Hi Phil, so far, I have added a setting "Fill Below All Skin" that forces the layer below a layer that has any amount of skin to be printed using zig-zag pattern (just like the existing Skin Edge Support Density feature). The zig-zag infill is shifted so that it it maximises the gap between it and the bumps on the layer below. As mentioned above, you will need to use a density for the zig-zag infill that is compatible with the pattern and density used for the infill below but that is easy to do by using the layer view to check that the zig-zag lines don't fall on the bumps.

[image]

Shall I make a release with this so you can try it out and see if it is adequate?

PhilBaz, 2025-04-06:
Hi Mark !

Apologies for the delay. We had an epic 1 in 30 year ice-storm in my part of Canada, and it took out the power at our semi-rural house for 6 days. 

Quite the experience. 

Yes! Very excited to try this out. I saw there was a new release.

If possible for a future release my hope is you will consider adding speed/flow/etc, for this new line type. 

But then I haven't checked the notes for the release, so perhaps you already have.

Thanks again for all your work on the best slicer out there! :)

smartavionics, 2025-04-06:
Wow, I hope you didn't suffer too much damage.
In the UK we're now getting once in a decade rainfall events every year! Weather's gone super weird.
Give it a try and let me know what extra stuff that you can't live without needs adding and I'll take a look.